slimbootloader/slimbootloader #2060: [EHL] FPT tool not working properly

sharkblue2009, 2023-10-23:
Hello,

We had try FPT tool in the CSE package(Intel_CSE_15.40.30.2879 V2) on Linux. 

Seems some blocks has erase error.
I had add below code to disable bios protect in stage2 code:
    Fspscfg->PchLockDownBiosInterface = FALSE;
    Fspscfg->PchLockDownBiosLock = 0;
and disabled "ProgramSecuritySetting()"

the FPT output is below:

----------------------------------
_/home/test# ./FPT -f outimage.bin

Intel (R) Flash Programming Tool Version: 15.40.30.2889
Copyright (C) 2005 - 2023, Intel Corporation. All rights reserved.

Reading HSFSTS register... Flash Descriptor: Valid

    --- Flash Devices Found ---
    ID:0xEF4019    Size: 32768KB (262144Kb)

 Region does not exist.

Processing Flash memory block  5644 from  8191.
Trying to erase the same block (iteration: 2)
Trying to erase the same block (iteration: 3)

Warning: Failed to erase this block 3 times!!

Error 185: FCERR is set. Hardware sequencing failed. Make sure that you have access to target flash area.

FPT Operation Failed._

-----------------------------------------

the debug FSP output is :

------------------ PCH Lock Down Config ------------------
 GlobalSmi= 1
 BiosInterface= 0
 BiosLock= 0
 UnlockGpioPads= 0
------------------ PCH Flash Protection Config ------------------
 WriteProtectionEnable[0] = 0
 ReadProtectionEnable[0]  = 0
 ProtectedRangeLimit[0]   = 1FFF
 ProtectedRangeBase[0]    = 1300
 WriteProtectionEnable[1] = 0
 ReadProtectionEnable[1]  = 0
 ProtectedRangeLimit[1]   = 0
 ProtectedRangeBase[1]    = 0
 WriteProtectionEnable[2] = 0
 ReadProtectionEnable[2]  = 0
 ProtectedRangeLimit[2]   = 0
 ProtectedRangeBase[2]    = 0
 WriteProtectionEnable[3] = 0
 ReadProtectionEnable[3]  = 0
 ProtectedRangeLimit[3]   = 0
 ProtectedRangeBase[3]    = 0
 WriteProtectionEnable[4] = 0
 ReadProtectionEnable[4]  = 0
 ProtectedRangeLimit[4]   = 0
 ProtectedRangeBase[4]    = 0

Could you please check if we missed anything? How to get FPT tool work?

stanleyintel, 2023-10-24:
Hi Sharkblue,

   Please check Test ID SPI_07 in RDC# 617151 to know how to get FTP tool work.

sharkblue2009, 2023-10-24:
Hi Stanley,

From the Debug FSP log, if the SPI flash write protection was sucessful disabled? I just want to make sure.

I am following the instructions from "System Tools - Intel® Converged Security Engine Firmware 15.40 User guide" to run FPT.
I had look through the SPI_07 as you suggested.

by the log,  the FPT fails on specific block 5644 when erase,  if its conflict with "CSE manufacturing mode"?

ps: "mtdinfo /dev/mtd0" still show the device is "writeable: false"

stanleyintel, 2023-10-24:
Hi Sharkblue,

   You may misunderstand about the SPI flash write protection and FPT.

   For SPI Flash Write Protection: Intel does not recommend any customer to disable it. Let me repeat what Maurice mentioned in [a related thread first:](https://groups.io/g/slimbootloader/topic/82351644#465).

> For security concerns, SBL will lock down SPI flash write-protect at the end of Stage2 for normal boot flow.
> And as a result, it prevents any SPI write access from OS or application. So flashrom won't work by default.
> 
> However, if you fully understand the security risk, or the platform is used in a debug / testing environment, and you want to allow runtime SPI write in OS, it is also possible to disable SPI write protection in SBL. But it is not recommended in production since any malware in OS might be able to write to the flash.
> 

  For FPT: FPT is to check the design of Flash Descriptor Override (per board).

stanleyintel, 2023-10-24:
For further questions related to FPT, please file an IPS case, because FPT is not released by SBL project and the mix use of FPT with PchLockDownBiosInterface/PchLockDownBiosLock is not a standard use case so you may need some help from CSE team via IPS case.

sharkblue2009, 2023-10-24:
well, thanks a lot!

closed 2023-10-24 by sharkblue2009