slimbootloader/slimbootloader #2118: SBL fails to reboot on TigerLake

fridgemagnet3, 2024-01-22:
SBL fails to reboot TGL (using UP XTREME) after initial programming, instead it sits indefinitely at _FSP Requested Reboot_. A manual power cycle is required then in order to recover, at which point everything is ok.

This used to work and I think this was triggered as a result of [this change](https://github.com/slimbootloader/slimbootloader/pull/1825) which passes the FSP Reset status through _FspResetHandler_ and ultimately through to _ResetSystem_. However in the default _ResetSystemLib_, there is no handler for _EfiResetPlatformSpecific_, hence the board does nothing.

stanleyintel, 2024-01-23:
Thanks for reporting the issue. 

Correct, it is caused by no handler for EfiResetPlatformSpecific. May you cherry pick https://github.com/slimbootloader/slimbootloader/pull/2120/files and check if the patch can resolve your issue?

stanleyintel, 2024-01-25:
Hi @fridgemagnet3 

   Can you help to check if the patch can solve the issue? If not, please attach the boot console log.

fridgemagnet3, 2024-01-25:
Hi @stanleyintel 

Sorry I've been tied up with a few other things, it's on my list of things to do tomorrow. Hopefully I'll report back then.

Thanks. Jon.

stanleyintel, 2024-01-26:
Thank you. Will move the PR to review.

randylintw, 2024-05-28:
PR merged. Close it due to a prolonged period of inactivity.

closed 2024-05-28 by randylintw