slimbootloader/slimbootloader #2151: [EHL] How to write some codes to do something during S0 into S5

wt-kevin, 2024-03-08:
As object,

We want to do something during S0 into S5,
Does SBL offer the path for us?
if yes, could you offer process/path name?

stanleyintel, 2024-03-11:
Hello Kevin,

  Suppose you're discussing "do something" when an OS is running. Typically, you need to create a kernel driver to register the system's suspend event. For Linux, you can refer to this link https://www.kernel.org/doc/html/latest/driver-api/pm/notifiers.html to learn how do do it. I can also find a similar discussion for Windows on StackOverflow.

  If the "do something" is straightforward, e.g., writing a post code to a 7-segment display, and your OS supports ACPI, you can alter the _PTS method in ACPI. For EHL, you can find the method [here](https://github.com/slimbootloader/slimbootloader/blob/fc11630e3b9112851da41498344052abd6365032/Platform/ElkhartlakeBoardPkg/AcpiTables/Dsdt/Platform.asl#L179). For more details about the _PTS, please refer to the ACPI specification.

wt-kevin, 2024-03-11:
Hi Stanley,

We don't want to do something in ASL code, _PTS is for ASL code to do something, right?
We want to do something in BIOS code when pressing power off button(S0 -> S5) without any OS.
when pressing power button off under DOS or UEFI(no OS), we can do something in BIOS code via SMI or other? if yes, how to do?

stanleyintel, 2024-03-11:
Hello Kevin,

   When you press the power button in SBL stage1A/1B/2 or its payload, the board will power off. There's no interrupt handler in the default SBL to stop this from happening.

  About SMI: The SMI handler is not a solution to your question. It's somehow like a callback function that activates when a specific SMI event happens. If you want to enable SMI handler for a S5 event, you will need to: (1) implement a SMI handler; (2) register it; (3) implement a suspend event notifier in OS (you cannot avoid it); and (4) trigger your SMI handler when your notifier is called. The step 3 and 4 can be implemented in _PTS.

  In SBL, the support of SMI depends on the payload. The OsLoader currently doesn't support SMI, while UEFI Payload indeed supports SMI. You can find an example [here](https://github.com/tianocore/edk2/blob/ccf91b518f22102d446f26320110d30ea0fc1fa9/UefiPayloadPkg/BlSupportSmm/BlSupportSmm.c#L409). If you have more questions about UEFI SMI support or how to implement/register such handler, please post them in the EDK2 forum: https://edk2.groups.io/g/devel/messages.

wt-kevin, 2024-03-22:
hi Stanley,

May I know how to post question in EDK2 as below.
[image]

stanleyintel, 2024-03-22:
Hello Kevin,

  I hope you understand that your question falls outside the topics covered by our forum.

randylintw, 2024-06-06:
Kindly close it due to a prolonged period of inactivity.

closed 2024-06-06 by randylintw