slimbootloader/slimbootloader #2162: [MTL] build fail on Ubuntu

rafaelhuang-intel, 2024-04-03:
Hi,

I've build fail for MTL, below are fail log.

make: Nothing to be done for 'tbuild'.
Trim --source-code -l -o /home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii /home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iii
"iasl" -ve -oi -p/home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.aml /home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii
/home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  11997:   External (VMDE)
Error    6117 -                                                                                                        Existing object has invalid type for Scope operator ^  (VMDE [Untyped])

/home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  18774:         External(\_SB.PC00.RP12.CLID)
Error    6117 -                                                                                                                            Existing object has invalid type for Scope operator ^  (\_SB.PC00.RP12.CLID [Untyped])

/home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  20721:     External (\_SB.PC00.TDM0)
Error    6117 -                                                                                                                    Existing object has invalid type for Scope operator ^  (\_SB.PC00.TDM0 [Device])

/home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii  20722:     External (\_SB.PC00.TDM1)
Error    6117 -                                                                                                                    Existing object has invalid type for Scope operator ^  (\_SB.PC00.TDM1 [Device])

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20190509
Copyright (c) 2000 - 2019 Intel Corporation

ASL Input:     /home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.iiii -  487644 bytes   9592 keywords  25520 source lines

Compilation failed. 4 Errors, 1 Warnings, 2654 Remarks
No AML files were generated due to compiler error(s)
cp -p -f /home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.aml /home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.amli
cp: cannot stat '/home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.aml': No such file or directory
make: *** [GNUmakefile:563: /home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables/OUTPUT/Dsdt/Dsdt.aml] Error 1

build.py...
 : error 7000: Failed to execute command
        make tbuild [/home/intel/slimboot/slimbootloader/Build/BootloaderCorePkg/DEBUG_GCC5/IA32/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables]

build.py...
 : error F002: Failed to build module
        /home/intel/slimboot/slimbootloader/Platform/MeteorlakeBoardPkg/AcpiTables/AcpiTables.inf [IA32, GCC5, DEBUG]

- Failed -
Build end time: 13:46:06, Apr.03 2024
Build total time: 00:00:08

Error in running process:
  build --platform BootloaderCorePkg/BootloaderCorePkg.dsc -b DEBUG --arch IA32 --tagname GCC5 -n 8 -y Report.log -Y PCD -Y FLASH -Y LIBRARY

rafaelhuang-intel, 2024-04-03:
code already up-to-date

intel@intel-host-4790:~/slimboot/slimbootloader$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/Testing
  remotes/origin/bejean/SblOpen/Fsp24SupportRebase
  remotes/origin/eps_heci
  remotes/origin/master
  remotes/origin/revert-1736-sgx_icxd_fix
  remotes/origin/revert-1747-sgx_icxd_fix
  remotes/origin/revert-1965-ci_iasl_update
  remotes/origin/revert-1969-revert-1965-ci_iasl_update
intel@intel-host-4790:~/slimboot/slimbootloader$ git log
commit 6adb9072d548214dc895cd8a10065daf2cfe41b5 (HEAD -> master, origin/master, origin/HEAD)
Author: Yi Li <yi1.li@intel.com>
Date:   Fri Mar 29 14:41:31 2024 +0800

    Let Stage1B run in memory accoss all stages in fast boot mode

    This will increase time of Board PostMemoryInit hook and TempRamExit,
    but improve the performance of PostTempRamExit hook, PreSiliconInit hook
    and FSP-S.

    In total, the boot time can be reduced by ~20ms.

    Signed-off-by: Yi Li <yi1.li@intel.com>

randylintw, 2024-04-03:
I remember it could cause by acpi tool on Ubuntu. 
https://github.com/slimbootloader/slimbootloader/pull/2156

rafaelhuang-intel, 2024-04-07:
Ubuntu20.04:   >>  build fail

Checking Toolchain Versions...
- /usr/bin/python3: Version 3.8.10 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 1.1.1f (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.14.02 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20190509 (>= 20190509) [PASS]
- /usr/bin/git: Version 2.25.1 (>= 2.20.0) [PASS]
- /usr/bin/gcc: Version 9.4.0 (>= 7.3) [PASS]
...Done!

change to Ubuntu 22.04:   >> build Pass

Checking Toolchain Versions...
- /usr/bin/python3: Version 3.10.12 (>= 3.6.0) [PASS]
- /usr/bin/openssl: Version 3.0.2 (>= 1.1.0g) [PASS]
- /usr/bin/nasm: Version 2.15.05 (>= 2.12.02) [PASS]
- /usr/bin/iasl: Version 20200925 (>= 20190509) [PASS]
- /usr/bin/git: Version 2.34.1 (>= 2.20.0) [PASS]
- /usr/bin/gcc: Version 11.4.0 (>= 7.3) [PASS]
...Done!

it's seems MTL may need latest version of nasm, iasl & ACPI, so close this issue.

closed 2024-04-07 by rafaelhuang-intel