slimbootloader/slimbootloader #2172: How to reduce the size of SlimBootloader.bin?

ziv2013, 2024-05-02:
Hi,

I'm trying to porting slimboot to a ADLN-N platform(Latte Panda). And the original IFWI on it is 16MB.

After building with "BuildLoader.py build adln", I get a SlimBootloader.bin with 13312KB which is too big to 

put it with CSME.

I noticed the are Top Swap A/B and REDUNDANT A/B in it. How can I remove one?

Flash Map Information:
        +------------------------------------------------------------------------+
        |                              FLASH  MAP                                |
        |                         (RomSize = 0x00D00000)                         |
        +------------------------------------------------------------------------+
        |   NAME   |     OFFSET  (BASE)     |    SIZE    |         FLAGS         |
        +----------+------------------------+------------+-----------------------+
        +------------------------------------------------------------------------+
        |                               TOP SWAP A                               |
        +------------------------------------------------------------------------+
        |   SG1A   |  0xce5000(0xFFFE5000)  |  0x01b000  |  Uncompressed, TS_A   |
        |   ACM0   |  0xc80000(0xFFF80000)  |  0x065000  |  Uncompressed, TS_A   |
        +------------------------------------------------------------------------+
        |                               TOP SWAP B                               |
        +------------------------------------------------------------------------+
        |   SG1A   |  0xc65000(0xFFF65000)  |  0x01b000  |  Uncompressed, TS_B   |
        |   ACM0   |  0xc00000(0xFFF00000)  |  0x065000  |  Uncompressed, TS_B   |
        +------------------------------------------------------------------------+
        |                              REDUNDANT A                               |
        +------------------------------------------------------------------------+
        |   SG1B   |  0xa00000(0xFFD00000)  |  0x200000  |  Uncompressed, R_A    |
        |   KEYH   |  0x9ff000(0xFFCFF000)  |  0x001000  |  Uncompressed, R_A    |
        |   CNFG   |  0x9fb000(0xFFCFB000)  |  0x004000  |  Uncompressed, R_A    |
        |   FWUP   |  0x9db000(0xFFCDB000)  |  0x020000  |  Compressed  , R_A    |
        |   SG02   |  0x919000(0xFFC19000)  |  0x0c2000  |  Compressed  , R_A    |
        |   UCOD   |  0x82d000(0xFFB2D000)  |  0x0ec000  |  Uncompressed, R_A    |
        +------------------------------------------------------------------------+
        |                              REDUNDANT B                               |
        +------------------------------------------------------------------------+
        |   SG1B   |  0x62d000(0xFF92D000)  |  0x200000  |  Uncompressed, R_B    |
        |   KEYH   |  0x62c000(0xFF92C000)  |  0x001000  |  Uncompressed, R_B    |
        |   CNFG   |  0x628000(0xFF928000)  |  0x004000  |  Uncompressed, R_B    |
        |   FWUP   |  0x608000(0xFF908000)  |  0x020000  |  Compressed  , R_B    |
        |   SG02   |  0x546000(0xFF846000)  |  0x0c2000  |  Compressed  , R_B    |
        |   UCOD   |  0x45a000(0xFF75A000)  |  0x0ec000  |  Uncompressed, R_B    |
        +------------------------------------------------------------------------+
        |                             NON REDUNDANT                              |
        +------------------------------------------------------------------------+
        |   PYLD   |  0x42a000(0xFF72A000)  |  0x030000  |  Compressed  ,  NR    |
        |   UVAR   |  0x429000(0xFF729000)  |  0x001000  |  Uncompressed,  NR    |
        |   EPLD   |  0x239000(0xFF539000)  |  0x1f0000  |  Uncompressed,  NR    |
        |   MRCD   |  0x229000(0xFF529000)  |  0x010000  |  Uncompressed,  NR    |
        |   VARS   |  0x227000(0xFF527000)  |  0x002000  |  Uncompressed,  NR    |
        |   IPFW   |  0x226000(0xFF526000)  |  0x001000  |  Uncompressed,  NR    |
        |   EMTY   |  0x09a000(0xFF39A000)  |  0x18c000  |  Uncompressed,  NR    |
        +------------------------------------------------------------------------+
        |                              NON VOLATILE                              |
        +------------------------------------------------------------------------+
        |   RSVD   |  0x099000(0xFF399000)  |  0x001000  |  Uncompressed,  NV    |
        |   EMTY   |  0x000000(0xFF300000)  |  0x099000  |  Uncompressed,  NV    |
        +----------+------------------------+------------+-----------------------+

Thanks a lot.

Ziv

gdong1, 2024-05-02:
Hi Ziv,
ADL-N SBL layout and components size are configured by Platform\AlderlakeBoardPkg\BoardConfigAdlN.py.
You could reduce components size to have less blank size for that region to have a smaller SBL binary.
The details SBL layout is generated in file Outputs\adln\SlimBootloader.txt. you could refer it to update config file.
It is possible to have a 8MB SBL image with SBL A/B support.
It is not preferred to SBL redundant support which is required for power-safe firmware update support.

Thanks,
Guo

ziv2013, 2024-05-06:
Thank, after modification, I can make a 10MB rom.

closed 2024-05-06 by ziv2013