slimbootloader/slimbootloader #2207: ADL/MTL/RPL  Incorrect RsvdSmbusAddressTablePtr and PchNumRsvdSmbusAddresses

marcjones-syspro, 2024-06-24:
It looks like bad default values are being added in ADL/MTL/RPL for RsvdSmbusAddressTablePtr and PchNumRsvdSmbusAddresses and passed to the FSP.  

``` yaml
  - PchNumRsvdSmbusAddresses :
      name         : Number of RsvdSmbusAddressTable.
      type         : EditNum, HEX, (0x0, 0xFF)
      help         : >
                     The number of elements in the RsvdSmbusAddressTable.
      length       : 0x0001
      value        : 0x04
  - RsvdSmbusAddressTablePtr :
      name         : Point of RsvdSmbusAddressTable
      type         : EditNum, HEX, (0x0, 0xFFFFFFFF)
      help         : >
                     Array of addresses reserved for non-ARP-capable SMBus devices.
      length       : 0x0004
      value        : 0xfffc3ff8
```

https://github.com/slimbootloader/slimbootloader/blob/33060efdc634015e6507f1036223f930e1595fe6/Platform/AlderlakeBoardPkg/CfgData/CfgData_Memory.yaml#L539C5-L539C29

The use of the field in TGL makes a lot more sense by pointing to the SMBus Address Table and it doesn't have the yaml for these fields. 

```c
  Fspmcfg->PchNumRsvdSmbusAddresses  = sizeof (MemCfgData->SmbusAddressTable) / sizeof (MemCfgData->SmbusAddressTable[0]);
  Fspmcfg->RsvdSmbusAddressTablePtr  = (UINT32) (UINTN)MemCfgData->SmbusAddressTable;
```

https://github.com/slimbootloader/slimbootloader/blob/33060efdc634015e6507f1036223f930e1595fe6/Platform/TigerlakeBoardPkg/Library/Stage1BBoardInitLib/Stage1BBoardInitLib.c#L379

stanleyintel, 2024-06-25:
Hi Marc Jones,

   Thanks for the suggestion, I will check the new design first.

stanleyintel, 2024-06-27:
Hello Marc Jones,

  Thank you for your input. We have updated the code to make the configuration of RsvdSmbusAddressTablePtr for ADL/MTL/RPL as user-friendly as it is for TGL. You can find the commit in the repository at [the link](https://github.com/slimbootloader/slimbootloader/commit/b81f41924d4619e9da56e12677ef552a888d9beb).

marcjones-syspro, 2024-06-27:
Thanks, I'm closing this as resolved.

closed 2024-06-27 by marcjones-syspro