slimbootloader/slimbootloader #2220: How to enable the FSP debugging message?

ziv2013, 2024-07-06:
Hi，

I'm working on an opensource ADL-N project now.  And I get below error message during power-on(I guess the error is assert in MRC).

My question is: how to enable debug message output of MRC? Thanks.

BOOT: BP0 
MODE: 0
BoardID: 0x07
PlatformName: AdlNLp5
Memory Init
Not Found Saved MRC Data!

CAR Base 0xFEF00000 (0xFFF00)
FSPM Stack Base=0xFEF16000, Size=0xE9F00
FSPM CfgData assignment
SkipMbpHob = 0x0
FusaConfigPreMem Status Unsupported
Call FspMemoryInit ... Device Error

ASSERT_EFI_ERROR (Status = Device Error)
ASSERT [Stage1B] c:\buildbs\sbl\BootloaderCorePkg\Stage1B\Stage1B.c(429): !(((INTN)(RETURN_STATUS)(Status)) < 0)

STAGE_1B: System halted!

Z.t

stanleyintel, 2024-07-08:
Hi Z.t,

  Please check RDC #773559 (Slim Bootloader Debug BKM). Note that the "copy" command and the source of copied file are important. If you need more details, we can talk about it at https://groups.io/g/slimbootloader/topics.

ziv2013, 2024-07-08:
OK， thank I will check this document.

closed 2024-07-08 by ziv2013