slimbootloader/slimbootloader #2221: Regarding SBL porting.

vinayakmspl, 2024-07-08:
Dear SBL community,
                                  I'm a newbie to SBL . I'm currently working on porting SBL to a custom board based on x6212re elkhart lake processor and I request community to guide me on the same. I would like to know what are the bare minimum changes to be made in SBL src code in order to observe boot prints,  even if it's not booting it's okay but I want to see some prints atleast on the console upon flashing SBL on board. Looking forward for support.

Best Regards,
Vinayak R. G- Embedded Software Engineer @ Mistral Solutions

stanleyintel, 2024-07-08:
Hi Vinayak,

   The question is discussed in https://groups.io/g/slimbootloader/topics.

vinayakmspl, 2024-07-08:
Thank you Stanley..

closed 2024-07-17 by randylintw