slimbootloader/slimbootloader #2361: Missing attribute "type : Combo" for the parameter "EnableCrasheMode" in Platform/CommonBoardPkg/CfgData/CfgData_Common.yaml

IlyaGure, 2025-01-09:
Hello,
It looks like that the attribute "type : Combo" for the parameter "EnableCrasheMode" is missing in Platform/CommonBoardPkg/CfgData/CfgData_Common.yaml
That prevents the parameter "EnableCrasheMode" to be displayed in Config Editor.

[image]

Thanks.
Ilya

randylintw, 2025-01-09:
Maybe you can help submit a patch to fix it?

IlyaGure, 2025-01-12:
Hello,
Attached a patch file for your review.
Thanks.
Ilya.

randylintw, 2025-01-13:
Would you like to have a pull request for this patch ? thanks ~

IlyaGure, 2025-01-13:
I have tried, but I don't have the permission to open new branch on this site or pull local branch to this repository.
In my opinion it'd better you do it, would you mind?

randylintw, 2025-01-14:
okey, you need to fork a repo in your name and push the change to that, then create a PR. 
https://github.com/slimbootloader/slimbootloader/pull/2364

randylintw, 2025-01-15:
merged.

closed 2025-01-15 by randylintw