slimbootloader/slimbootloader #2411: Booting VxWorks on TGL with SBL and UEFI Payload.

Mounika250900, 2025-03-12:
We are trying to boot VxWorks on TGL platform, we use Slim bootloader with UEFI payload (EDKII). Do we need to modify any configuration in Slim bootloader or UEFI to boot VxWorks.

stanleyintel, 2025-03-13:
Hello Mounika,

Could you please take a look at our conversation on groups.io? Additionally, we recommend using the UEFI Payload that comes with the Intel TGL SBL package.

Mounika250900, 2025-03-13:
Hi,
We tried the Intel-released MR8 UEFI payload as you suggested, but we still couldn't get VxWorks to boot. We also discussed with the OS vendor about this. However, we were able to boot VxWorks with the American Megatrends UEFI; the booting problem we're having relates to the Slim Bootloader.

stanleyintel, 2025-03-13:
We're unable to provide further assistance with third-party OS. However, several customers have successfully managed to boot VxWorks by collaborating with the OSV, such as https://groups.io/g/slimbootloader/message/4308. If you need more help and the OSV is unable to assist, you might consider reaching out to these ISV for paid support.

randylintw, 2025-03-17:
Close this issue due to the suggestion provided.

closed 2025-03-17 by randylintw