slimbootloader/slimbootloader #2505: ADLN: how to make UARTs (COM ports) to work in Windows 11

molejnik-bct, 2025-08-13:
Hi,
I compiled Slimboot with EDK2 to boot Windows 11, however none of the 3 UART ports work correctly in Windows 11. All 3 UART ports are visible in Device Manager but show error code 10: The device cannot start. The UARTS were configured with UART mode 3 to see them all in Device Manager. When I used Uart Mode 1 then only one COM port was visible, but still not working, with error code 10.

I found there was a fix for MTL in commit c2f26c93ec5 ([MTL] UARTn Device Mode can't set to SerialIoUartCom) claiming all 3 UARTs/COM ports can be seen (and working properly?) in Win 11. I tried to apply the fix to Alder Lake N, but that did not resolve the issue - still the COM ports are in error state 10. Does Alder Lake N need the same/similar fix as MTL for UART mode 3? If so, what is the correct address of 'PchNvs->UC0[Index]' for Uart Mode 3 ? I tried (PCH_SERIAL_IO_BASE_ADDRESS + 0xD000 + (0x2000 * Index)) and also (PCH_SERIAL_IO_BASE_ADDRESS + 0x14000 + (0x2000 * Index)) without success.

The UART ports itself do work correctly when used with Ubuntu 24.04 and when UART mode is set to 1 - so I believe it is not a hardware issue, but rather a software configuration issue possibly fixable in Slimboot code.

BTW. I installed the Intel Chipset drivers and Intel Serial IO driver as well, but that did not fix the issue.

Any idea how to make the COM ports to work in Windows 11 (to resolve Error code 10) on Alder Lake N platform? Thanks.

stanleyintel, 2025-08-14:
Thanks for the detailed report. Based on your description, the "Communication Port (COM1)" showing error code 10 in Device Manager is likely the UART from the SIO controller (IT8659), not the PCH UART. Since ADL-N Slim Bootloader does not enable IT8659 by default, this results in the yellow bang. You can suppress this by setting in AcpiPlatform.c: PlatformNvs->PcdIT8659COM = 0.

The actual PCH UART devices appear under System Devices in Device Manager. To enable them, you need to resolve resource conflicts, e.g., UART2 shares GPIO pins with CNVI (refer to the ADL-N GPIO table).

I’ve created a test patch that demonstrates how to enable UART0–2 on ADL-N. While I haven’t verified hardware functionality, it should help guide your configuration for Windows: https://github.com/stanleyintel/slimbootloader/commit/f80dabab2224e95d3333a7afde5a25d0aedde8d6

Lastly, regarding to the the error code 10 for IT8659: since it’s not currently a priority for us, we may enable it for ADL-CRB on a best-effort basis when time permits.

molejnik-bct, 2025-08-14:
Hi Stanley,
thank you for the quick response and and a solution that works. I applied your fixes and now all 3 COM ports work correctly in Windows 11. I had to set UART mode on all three ports to value 3 (not 1 as you indicated) and then do the address translation when UART mode 3 was used:
`PchNvs->UC0[Index] = (UINT64)(PCH_SERIAL_IO_BASE_ADDRESS + 0x1F000 + (0x2000 * Index));`
The IRQ settings were required as well, but I had them already fixed.
The CNVI in my setup was also already turned off (as tested in Linux where the UARTS worked fine in UART Mode 1), so all 3 UARTs started to work out of the bat in Windows 11.

Please consider the issue fixed, feel free to close this issue. Thank you.

[image]

molejnik-bct, 2025-08-14:
Actually, I have a related question to UARTs. I can see these 3 UARTs only if I disable the debug serial port in BoardConfig.py:
self.DEBUG_PORT_NUMBER = 0xFF
If I use "self.DEBUG_PORT_NUMBER = 0" then only 2 UART ports are visible. I understand the UART0 is forced to UART Mode 4 and is not exposed as COM port if  "self.DEBUG_PORT_NUMBER = 0". I'm happy to use "self.DEBUG_PORT_NUMBER = 0xFF" but that poses an issue with with EDK2. Basically it seems that EDK2 tries to print messages (its menu options) to such 'invalid' debug port number (0xFF) and it freezes. Therefore it is not possible to operate EDK2 at all - impossible to set any EDK2 menu options. My question is: Is there a way to make DEBUG_PORT_NUMBER 0xFF to work with EDK2 ? Or is there a build option that disables all debug logging for EDK2? I know it is a bit out of topic, but if there is a solution it would much appreciated. Thank you.